Makuna/DFMiniMp3 #3: The example sketches need SoftwareSerial-usage hints

AustinSpafford, 2017-04-08:
The examples currently fail to compile for an Arduino Uno due to it only having one hardware serial port, leading to this error:

```
'Serial1' was not declared in this scope
```

I recommend replacing the line referencing "Serial1" in the examples with the following set of lines:

```cpp
DFMiniMp3<HardwareSerial, Mp3Notify> mp3(Serial1); // If this line fails to compile, try replacing it the two with the two lines below. This comes up because some arduino boards only have one hardware serial port, so a software serial port is needed instead.
//SoftwareSerial secondarySerial(10, 11); // RX, TX
//DFMiniMp3<SoftwareSerial, Mp3Notify> mp3(secondarySerial);
```

Makuna, 2017-04-09:
https://github.com/Makuna/DFMiniMp3/pull/4

Makuna, 2017-04-09:
done

closed 2017-04-09 by Makuna