Makuna/DFMiniMp3 #6: How to play next song on OnPlayFinished

stigi, 2018-02-17:
I am wondering if there's a way to call `next()` or any other playback method from a notification callback. All methods of `T_NOTIFICATION_METHOD` are declared static and can't access outside state.
The only way I could imagine is to change the API so that each of the static callbacks gets a reference passed. Some kind of context it could use to trigger additional logic.

Am I missing currently available possibilities with the API?

Thanks.

Makuna, 2018-02-18:
Most Arduino projects will define the primary object as a global.  The static method can call this global.

xfjx, 2018-03-10:
Can you please be a little bit more verbose. I'm running into the same problem right now. Even when declaring my mp3 Object as static I can not access it from the notification class.

Thank you very much. I'm using you library in a jukebox for kids. It is much better than the original library from dfrobot.

stigi, 2018-03-10:
Btw, I *forked* the code into my project and tried out the context variant.
In case anyone wants to check it out https://github.com/stigi/fritz-s-box

closed 2018-03-10 by stigi