Makuna/DFMiniMp3 #13: Error after looping a few times

9H1LO, 2019-02-25:
running on ESP8166 using software serial

After a couple of repeat plays of the same track i get:
17:08:21.896 -> Com Error 255
17:08:29.527 -> 
17:08:29.527 -> Play finished for #40
17:08:37.174 -> 
17:08:37.174 -> Play finished for #40
17:08:44.823 -> 
17:08:44.823 -> Play finished for #40

playback continues but it seems to get stuck in the notification class and doesn't keep the main loop running

Makuna, 2019-02-25:
Please provide more information specific to your problem
* IDE version.
* Esp8266 board support version.
* DFMinimMp3 Library version
Also provide Minimal Sample sketch that demonstrates the problem.

To me it sounds like your connections maybe faulty causing communications errors.  It uses the normal serial timeout waiting for response data, if it doesn't get it then it seems like the main loop isn't running; but it will return to it after the serial times out.

9H1LO, 2019-02-25:
* IDE is arduino 1.8.8 
* ESP8266 2.4.2
* DFMinimMp3 Library 1.0.2
* board is wemos d1 mini

Sketch is sample from library with some modifications to do loop play and read current state/track/volume every few seconds

yes it seems like the comms is giving errors or being lost, maybe faulty mp3 module or issue with esp pins ? doing init as follows:
SoftwareSerial SerialMP3(12, 14, false, 512); //mp3 // RX, TX
DFMiniMp3<SoftwareSerial, Mp3Notify> mp3(SerialMP3);

have you found certain pins better for serial?

it works fine for a while, but then just gives the error at random, especially noted when looping through short mp3 files, i.e. lots of "play finishes"

Makuna, 2019-02-25:
Without the actual sketch you are using, and the mp3 files, I can't reproduce.

You might try GPIO4 and GPIO5; as they aren't used by other internal hardware so less likely to conflict.

I would have thought It might be all the constant reading of data is overloading the software serial library send/recieve buffers before the data can be removed by either the device or the Esp8266.  But you increased the recieve buffer to 512.  You might try removing the extra calls and see how that goes.

9H1LO, 2019-02-25:
i will try that

after experimenting some more, it seems ( i think ) the issue only occurs a while after calling a getStatus()  during playback, maybe the player doesn't like lots of comms going on whilst playing ?

Makuna, 2019-02-25:
In https://github.com/Makuna/DFMiniMp3/pull/14 I improved the packet decode errors.
If you grab the github version of the library (see readme, its not to hard) you can try this and see if the new errors help.

9H1LO, 2019-02-27:
i did as you suggested and it improved a lot, but still gives issues
the main loop calls waitMilliseconds(5000), if i don't do the call for getCurrentTrack() it runs fine
on error it just seems to get keep in the notification callback and doesn't exit to the main loop

`void waitMilliseconds(uint16_t msWait){
   static unsigned long timer = millis();
  if (millis() - timer > msWait) {
    iter = iter + 1;
    Serial.print("Iteration: ");
    Serial.println(iter);
   timer = millis();
   mp3.loop();
   getVolume();
   Serial.print("Volume: ");
   Serial.println(getVolume());
   playerTrack = mp3.getCurrentTrack();
   Serial.print("Track: ");
   Serial.println(playerTrack);
   delay(1);
  }
}`

here is when it happens:

18:24:11.584 -> Iteration: 1667
18:24:11.617 -> Volume: 2
18:24:11.754 -> Track: 41
18:24:16.575 -> Iteration: 1668
18:24:16.608 -> Volume: 2
18:24:16.743 -> Track: 41
18:24:21.600 -> Iteration: 1669
18:24:21.600 -> Volume: 2
18:24:21.736 -> Track: 41
18:24:26.604 -> Iteration: 1670
18:24:26.604 -> Volume: 
18:30:11.971 -> Play finished for #41
18:36:46.166 -> 
18:36:46.166 -> Play finished for #41
18:43:20.347 -> 
18:43:20.347 -> Play finished for #41
18:49:54.550 -> 
18:49:54.550 -> Play finished for #41
18:56:28.729 -> 
18:56:28.729 -> Play finished for #41

Makuna, 2019-02-27:
You modified the waitMilliseconds and it no longer functions the way it supposed to from the examples.  

Please provide the full MINIMUM Sketch the can reproduce the problem.  Without that, I can only assume its a problem in your code and I will be forced to close this issue; as issues are about problems in THIS code base and not questions and support forum.

Makuna, 2019-03-30:
closed due to lack of response for a month.

closed 2019-03-30 by Makuna