Makuna/DFMiniMp3 #20: Invalid conversion in lib

fronbasal, 2019-04-26:
Hello! I am attempting to run the "random" example, yet it is failing (`invalid conversion from 'uint16_t {aka short unsigned int}' to 'DfMp3_PlaybackMode' [-fpermissive]`).
Here is the full stack trace:
```
/home/fronbasal/Arduino/libraries/DFPlayer_Mini_Mp3_by_Makuna/src/DFMiniMp3.h: In instantiation of 'DfMp3_PlaybackMode DFMiniMp3<T_SERIAL_METHOD, T_NOTIFICATION_METHOD>::getPlaybackMode() [with T_SERIAL_METHOD = HardwareSerial; T_NOTIFICATION_METHOD = Mp3Notify]':
/home/fronbasal/Desktop/PlayMp3/PlayMp3.ino:81:39:   required from here
/home/fronbasal/Arduino/libraries/DFPlayer_Mini_Mp3_by_Makuna/src/DFMiniMp3.h:184:35: error: invalid conversion from 'uint16_t {aka short unsigned int}' to 'DfMp3_PlaybackMode' [-fpermissive]
         return listenForReply(0x45);
                                   ^
exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).
```

Makuna, 2019-04-27:
https://github.com/Makuna/DFMiniMp3/pull/21

fronbasal, 2019-04-27:
Thank you for the quick fix!

closed 2019-04-27 by fronbasal