Makuna/DFMiniMp3 #23: Documentation for Mp3Notify class

TheeCaptain, 2019-05-05:
I cannot find documentation for the Mp3Notify class. I was hoping to use OnPlayFinished() to know when my track has completed playing.

I would like to clear my isPlaying variable when the track is done. Because there is no documentation for the Mp3Notify callbacks, I don't know if they are not working properly or if I have just assumed in error what they do.

Thank you for your time and the great library.

Makuna, 2019-05-06:
https://github.com/Makuna/DFMiniMp3/wiki/Notification-Method

closed 2019-05-06 by Makuna

TheeCaptain, 2019-05-06:
Terrific, thank you. I don't know why I didn't see this before. Much appreciated.

Makuna, 2019-05-07:
That's because I just added it today ;-)

xXTobi91Xx, 2019-06-24:
Hi, thank you for your help.
I am fairly new to programming and i was hoping if you could provide an example code how to call static void OnPlayFinished(uint16_t track) in the main loop to receive the track number after a song was played.

Thank you very much

Makuna, 2019-06-24:
@xXTobi91Xx Asking questions in an  issue that is closed is not a valid way to ask questions.  Issues are not a valid place to ask questions in general.  There is a gitter channel for that.  See the readme or wiki.  Further, there are examples already for how to implement the callback.