Makuna/DFMiniMp3 #24: Request for comments: Remove SerialMethod Template?

lukx, 2019-05-07:
Hey,
the template coding is kind of nice, however I feel like it's keeping me from properly injecting an instance of the DfMiniMp3 class into another class, since I'd have to carry around the same template just for the sake of it. I can't just do the following:

```
class PlaybackController {
  private:
     DfMiniMp3 & mp3player;
}
```

Since the usage of the Serial is mostly "begin", "timeout" and then the properties of "stream", what if instead of using the templated serial class, we re-write to accepting an instance of "Stream", which would then have to be pre-instantiated?

I see that this would be a breaking change, but still hope to hear your thoughts?

Pro: 
- easier passing of class references
- less dependencies or required knowledge in the interface

con
- additional step to setup the serial port
- breaking change

```
template<class T_NOTIFICATION_METHOD> class DFMiniMp3
{
public:
    DFMiniMp3(Stream & serial) :
        _serial(serial),
        _lastSendSpace(c_msSendSpace),
        _isOnline(true)
    {
```

Makuna, 2019-05-07:
Why isn't your class using a template for mp3 type? ;-)

There are more cons to your solution than you think; most are just not obvious.  

The normal way to accomplish what you want, is to use typedef, like...
```
typedef DFMiniMp3<HardwareSerial, Mp3Notify> Mp3Device;
```
Then use `Mp3Device` as you want.

closed 2019-05-11 by Makuna

Modifikator, 2022-05-12:
Hi. I need to place the mp3player instance in a structure that is in a vector. How to do it right?
 (I have little experience in c++)

 `
  #include <SoftwareSerial.h>
  #include <DFMiniMp3.h>

  typedef DFMiniMp3<SoftwareSerial, Mp3Notify> DfMp3;

  struct MD {
      bool log;  
      bool ws;
      time_t t;
      String name;
      SoftwareSerial *com; 
      DfMp3 *device;         
  };

  std::vector<MD> myDevice;

void initDFPLAER(String name, uint8_t RX, uint8_t TX, bool _ws, bool _log) {

      MD mp3;
      mp3.log = _log;
      mp3.ws = _ws;
      mp3.t = now(); 
      mp3.name = name;   
      mp3.com = new SoftwareSerial(RX, TX);
      mp3.device = new DfMp3( mp3.com ) ;

    myDevice.push_back(mp3);     

    myDevice[0].device ->begin();

    myDevice[0].device->playMp3FolderTrack(1); 

}

void waitMilliseconds(uint16_t msWait)
{
  uint32_t start = millis();  
  while ((millis() - start) < msWait)
  {
   myDevice[0].device ->loop();  
    delay(1);
  }
}

loop() {
    waitMilliseconds(1000);
}
`

Makuna, 2022-05-12:
@Modifikator  asking questions on a closed issue is not the way to get help. Either use the [discussions area] (https://github.com/Makuna/DFMiniMp3/discussions) or the gitter channel.