Makuna/DFMiniMp3 #25: getTotalFolderCount returns total number of files

lyruiot, 2019-06-16:
I am using an Arduino Nano and DFPlayer aka MP3-TF-16P

in my setup function, I call getTotalFolderCount ();

Serial.print("Folders in SD \t"); Serial.print(mp3.getTotalFolderCount( ));Serial.print("\n");
Serial.print("Tracks in folders **\t"); Serial.print(mp3.getTotalTrackCount ( ));Serial.print("\n");
Serial.print("Tracks in folder 01\t"); Serial.print(mp3.getFolderTrackCount(1));Serial.print("\n");
Serial.print("Tracks in folder 02\t"); Serial.print(mp3.getFolderTrackCount(2));Serial.print("\n");
Serial.print("Tracks in folder 03\t"); Serial.print(mp3.getFolderTrackCount(3));Serial.print("\n");

Output is shown below:
Folders in SD 33
18:53:29.583 -> Tracks in folders ** 33
18:53:29.678 -> Tracks in folder 01 12
18:53:29.724 -> Tracks in folder 02 11
18:53:29.817 -> Tracks in folder 03 10

I think getTotalFolderCount (which inturn calls sendPacket(0x4F); ) counts the total number of tracks rather than the folders.

Has someone tried this command?

seisfeld, 2019-06-16:
Hmm this is odd. See PDF which i attached to Pull Request #8. This is from the original vendor Flyron and it states:

> 0x4F | Query total folder numbers of the storage device

I can only assume it depends on the exact chip your module has. There are many derivates and clones which have slightly different firmware and feature sets. I have one module here where `getTotalFolderCount()` works as intended and another one where it does not work at all.

What chip does your module have?

lyruiot, 2019-06-16:
Mine has a YX5200-24SS IC. 
All I am trying to achieve is to select consequitive folder upon press of a button. 
Is there any other means to achieve given getTotalFolderCount( ) doesnt work on my chip.

seisfeld, 2019-06-16:
Well you can switch folders no problem, you just can’t avoid selecting next when there is no next anymore. You’ll catch an error (after which you could then block the next function or something.

lyruiot, 2019-06-16:
https://github.com/myTonino/Tonino-Firmware/tree/master/Tonino

Is this the right one?

lyruiot, 2019-06-16:
[image]

Does the filename of the mp3 files really matter when counting the number of folder?

seisfeld, 2019-06-16:
We achieve the best result when the file name is a 3 digit zero padded number, like `001.mp3` while most chips also allow additional chars like `001 Track 1.mp3` or similar.

Makuna, 2019-06-17:
What does `getTotalTrackCount` return?
I believe the difference between these is that one returns the total tracks on the media, while the one you are calling returns the count of tracks IN FOLDERs. 

And yes, there are differences between chips, there is one API call I removed as non of the newer ones support it at all.

lyruiot, 2019-06-17:
Serial.print("Tracks in folders **\t"); Serial.print(mp3.getTotalTrackCount ( ));Serial.print("\n");
Out is shown below:
 18:53:29.583 -> Tracks in folders ** 33
There are totally 33 files in 3 folders in my case.

Makuna, 2019-06-17:
Place a mp3 file in the root.  Then call it.  You should see different numbers.

Makuna, 2020-05-19:
(stale)

closed 2020-05-19 by Makuna