Makuna/DFMiniMp3 #27: File/Format parser for DFplayer Mini

Schallbert, 2019-08-14:
I have written a script to have M3U playlist files parsed and the MP3 files copied into a folder structure, renamed in a way that it can be directly read by DFplayer mini.
It has nothing to do with your library but I thought it may come in handy.
Pls have a look at https://github.com/Schallbert/DFplayer_AutoStructureFromPlaylist
to see whether you can take advantage of this script.
Regards, 
Schallbert

Makuna, 2019-10-07:
If you want, add a link to your work in the Wiki [here](https://github.com/Makuna/DFMiniMp3/wiki#projects-that-use-this-library)

Schallbert, 2020-03-29:
Updated the Wiki as recommended. 
Win10 x64 .exe now also available on project.
Closing issue.

closed 2020-03-29 by Schallbert