Makuna/DFMiniMp3 #28: Endless loop in listenForReply

grimmpp, 2019-10-05:
When I call `mp3Player.getFolderTrackCount(folder);` for a folder value which is not valid. I get a response on `T_NOTIFICATION_METHOD::OnError(replyArg);` which is good but then the programm loops endless in the function `uint16_t listenForReply(uint8_t command)` because there is a condition `} while (command != 0);` which never can changes because the variable `command` never gets changed and is not `0` from the beginning.

Without properly knowing it I first thought it should be `} while (replyCommand != 0);`. It also eventually terminates but takes quite long until `replyCommand ` becomes `0`.

Does it make sense to directly exit in case on any notification?

Makuna, 2019-10-05:
Do you have a small sketch that duplicate the issue? 

There are two locations in that method where OnError is called, one will exit, one will continue to loop waiting for a response.  

The one that continues "may" need to be changed to ...
```
    case 0x40:
        T_NOTIFICATION_METHOD::OnError(replyArg);
	if (_serial.available() == 0)
	{
	    return 0;
	}
        break;
```

grimmpp, 2019-10-05:
for me its the first case:
```
if (readPacket(&replyCommand, &replyArg))
   {
   if (command != 0 && command == replyCommand)
      {
         return replyArg;
      }
      else
      {
         switch (replyCommand)
         {
...
            case 0x40:
               T_NOTIFICATION_METHOD::OnError(replyArg);
            break;
```

Here is a little ino which reproduces the behavior:
```
#include <DFMiniMp3.h>
#include <SoftwareSerial.h>

class Mp3Notify {
public:
  static void OnError(uint16_t errorCode) {
    // see DfMp3_Error for code meaning
    Serial.print("Com Error ");
    Serial.println(errorCode);
  }
  static void OnPlayFinished(uint16_t track) {
    Serial.print("Track beendet");
    Serial.println(track);
  }
  static void OnCardOnline(uint16_t code) {
    Serial.println(F("SD Karte online "));
  }
  static void OnCardInserted(uint16_t code) {
    Serial.println(F("SD Karte bereit "));
  }
  static void OnCardRemoved(uint16_t code) {
    Serial.println(F("SD Karte entfernt "));
  }
  static void OnUsbOnline(uint16_t code) {
      Serial.println(F("USB online "));
  }
  static void OnUsbInserted(uint16_t code) {
      Serial.println(F("USB bereit "));
  }
  static void OnUsbRemoved(uint16_t code) {
    Serial.println(F("USB entfernt "));
  }
};

SoftwareSerial mySoftwareSerial(2, 3); // RX, TX
static DFMiniMp3<SoftwareSerial, Mp3Notify> mp3(mySoftwareSerial);

void setup() {
  Serial.begin(115200);

  mp3.begin();
  mp3.setVolume(10);
}

void loop() {
  Serial.println("start loop");
  mp3.loop();

  // folder 1 does exist.
  Serial.print("get track count of folder 1: ");
  uint16_t numTracksInFolder = mp3.getFolderTrackCount(1);
  Serial.println(numTracksInFolder);

  // folder 5 does not exist.
  Serial.print("get track count of folder 5: ");
  numTracksInFolder = mp3.getFolderTrackCount(5);
  //should return "0"
  Serial.println(numTracksInFolder);

  Serial.println("end of loop");
}

/*
In case of an error program keeps hanging in lib DFMiniMp3 (line:  'numTracksInFolder = mp3.getFolderTrackCount(5);') and does not come back.

OUTPUT:
start loop
get track count of folder 1: 1
get track count of folder 5: Com Error 6
 */

```

Makuna, 2019-10-07:
https://github.com/Makuna/DFMiniMp3/pull/29

Makuna, 2019-10-07:
1.0.4 has the fix

closed 2019-10-07 by Makuna