Makuna/DFMiniMp3 #31: Wake up after sleep with setPlaybackSource

accidentalrebel, 2019-11-13:
Not exactly an issue but I've read your note in the wiki that says the only way you know how to wake up is using reset()

> Note: currently the only way to wake up the module that I know of is with reset(); but this is a slow process and may cause click noises in the audio output.

I learned from [this library](https://github.com/PowerBroker2/DFPlayerMini_Fast) that you can initiate a wake up by setting the play source to SD. In our case it can be done by calling:

>  setPlaybackSource(DfMp3_PlaySource_Sd);

I tried it and it worked. Just thought you should know.

Makuna, 2019-11-13:
You can go ahead and add that to the wiki.  I will give it a try.

Thanks

accidentalrebel, 2019-11-13:
Hi. I think I need to be added as contributor to have editing privileges on the wiki.

I can add my project too on the **Projects that use this library section.**

seisfeld, 2019-11-14:
I can confirm, the Wiki seems not editable.

Makuna, 2019-11-14:
For some reason it got switched to collaborators only. I believe that Wikis should be open and it always gives me the chance to revert what others do anyhow.  

It should now allow you to edit.

seisfeld, 2019-11-14:
Yap it’s editable now! Thanks @Makuna. 👍🏻

accidentalrebel, 2019-11-14:
I updated the wiki. Since I'm the only one who was confirmed that my approach is working I updated the note to say that we have two confirmed methods.

> Note:
There are two confirmed ways to wake up a module. Try any of the two and see which works best for your particular hardware:
>   * Calling reset(), although this is a slow process and may cause click noises in the audio output
>   * Calling setPlaybackSource(DfMp3_PlaySource_Sd)

If @Makuna  can confirm then that would be great.

ghmartin77, 2020-01-27:
Please note #37. Might have an impact on this one if fixed. I don't suppose it has, but should be checked, if the fix is applied, though.

Makuna, 2020-02-22:
New release exposes it.

closed 2020-02-22 by Makuna