Makuna/DFMiniMp3 #32: Request: LoopFolder 0x17

Iramoo, 2019-11-27:
Hi Makuna,
i'm using your nice libary for a children player. Now i've got the problem that the loop folder command (0x17 ) is not yet supported. Am I missing something or would it be possible to add this to the library. 

Thank you

Greetings Iramoo

Makuna, 2019-11-27:
Add this to the DfMiniMp3.h file of my library to see if it works.  Place it right after the loopGlobalTrack method.  Not all chips support it, unfortunately the ones I have do not so I can't test it.

```
    // sd:/##/*
    void loopFolder(uint8_t folder)
    {
        sendPacket(0x17, folder);
    }
```

Iramoo, 2019-11-27:
Wow very fast answer and solution. I'will try that and give a feedback how it works. 
Thank you.

Iramoo

Iramoo, 2019-12-22:
Update: As for now, i can say it works for me, thanks again. But i haven’t done “heavy” testing. 

Grettings  iramoo

Makuna, 2020-01-18:
release 1.0.5

closed 2020-01-18 by Makuna