Makuna/DFMiniMp3 #33: Detect DFPlayer brownout/crash

SciLor, 2019-12-12:
When a DFPlayer browns out either because of a too high volume or because of a too low supply current it goes nuts:
It starts to create a noise that sounds like a machine gun at full volume.
At the same time the player doesn't answer to serial requests anymore for some seconds. After that it resets, sends the 0x81 (129) error code and works fine again.

If the error code would be sent nearly immediately I could cut the dfplayers power supply / disable its amp (modified io pin).

Makuna, 2019-12-12:
This sounds like a hardware/firmware request.  This repo nor I have any relationship with the chip maker of which this issues is directed.

If it doesn't reply to serial, there is little that this library could do.  

Maybe add power management monitoring to your solution?

closed 2019-12-12 by Makuna

SciLor, 2019-12-12:
I am searching a brownout detection in software.
It would be possible to detect if the rx error would be thrown faster.

Makuna, 2019-12-12:
The default serial timeout is 20 seconds, the library already shortens it to 10 seconds.  This is due to how long the normal response can take with a little overhead; any shorter and it won't work in the normal cases.
The problem is the brown out can happen at any point, even in the middle of reading a packet.
After calling begin, you could try to call serial.setTimeout() (the one you use for the device) with a short value like 1000 (a second) and see how that works.   But again, this maybe too short for normal operation.

SciLor, 2019-12-12:
I see, there is any chance to detect this in software. I have switched from 3.3V to 5V power supply for the dfplayer and it seems to fix it. If it crashes it only resets, instead of making noises loudly.