Makuna/DFMiniMp3 #34: double invocation of OnPlayFinished()

Hannes-III, 2019-12-19:
First I thought there is something wrong when I discovered that the Method is called twice after my File teminated to play.
Then I read the documentation where it is clearly discribed that it can happen. I could have saved some time if I would have read it earlier. As always :-).
But now there is my question:
In the Callback method I start playing a new song and I increment a counter. As this method is called twice, the song is started twice. And my counter counts double.
Do you have any solution how one can find out if this is a Double invocation or not, so that I can return immediately?
Thank you.

seisfeld, 2019-12-19:
I solved it like this:

```
static void OnPlayFinished(uint16_t returnValue) {
  playNextTrack(returnValue);
}
```

`returnValue` gives you the global track number of the track just finished. This passes to the 
`playNextTrack()` function in which I do something like this:

```
void playNextTrack(uint16_t globalTrack) {
  static uint16_t lastCallTrack = 0;

  if (lastCallTrack == globalTracky) return;
  else lastCallTrack = globalTrack;

  // real code comes here
}
```

Which essentially skips the second call... You can check my [full code](https://github.com/seisfeld/TonUINO/blob/master/tonuino.ino) if you are interested.

Note: This does not work for callbacks happening due to adverts. In that case you better find a DFPlayer Mini without double callbacks. I have one from MH-ET LIVE here which does no fire twice... but ymmv.

Hannes-III, 2019-12-20:
I just analyzed the Serial Communication with the MP3 Player. At the end of a track the Player really sends two Completed play `0X7e 0Xff 0X06 0X3d 0X00 0X00 0X04 0Xfe 0Xba 0Xef` to the Arduino.
I Think there is nothing one can do about it. I am using the DFPlayer Mini from AZ-Delivery.
Your Solution, to compare last Track with current track is good if you can guarantee, that you do not play a song twice.

As there really is no way to solve this, all we can do is guess. My current solution is not bulletproof either, but it is a good guess:
'''
class Mp3Notify{
    public:
    static uint32_t LAST_FINISH;
    static void OnPlayFinished(uint16_t track){
    uint32_t now=millis();

    if(now - LAST_FINISH < 200){
        Serial.println("Skip double Invocation ("+String(now - LAST_FINISH)+" millis)");
        LAST_FINISH=now;
        return;
    }
    LAST_FINISH=now;

    Serial.print("Play finished for global# ");
    Serial.println(track);
    repeat(track);
    }
};
uint32_t Mp3Notify::LAST_FINISH=millis();
'''
Maybe the best way would be to combine both solutions. If the loop() hangs for long, my 200ms wont last.

btw:
> callbacks happening due to adverts
On my Player, when playing an advert `mp3.playAdvertisement(4)` the Callback method never gets called.

I close this now as there will never be a solution.

closed 2019-12-20 by Hannes-III