Makuna/DFMiniMp3 #35: make DFMiniMp3 compilable in platformio (change private and public order)

bridystone, 2020-01-16:
Hi @Makuna ,

I have an hopefully easy request to you regarding your great library.

Can you please change the order of the public and private declarations.
It would make the Library compilable in straight forward compilers.

in the current setup the compiler complaints that it does not know the private variables when being used in the public area - Since they are declared further down...

most c compilers are dumb like this to enable fast processing.

SOLUTION:
if you would first declare your private section and later declare the public stuff, it would know the private components.
(I've already tested it in my project.)

With this solution all developers could easily use your library with any compiler.

Thanks in advance
Jens

Makuna, 2020-01-16:
There is something more going on here.  C++ doesn't care the order, and it is common practice to public at the top, then protected, then private.  AND I do not know a single compiler that cares either way.

Now if there is a pre-processor script running I might agree they would be simple; but the compilers are fine.

I have other libraries that follow this common standards that work in platform IO, so I really doubt this is the issue you are seeing.  There must be another issue.

I have you contacted the platformIO teams about what  you are seeing?  They should fix that.

proddy, 2020-01-16:
I agree, this is not a compiler issue. Share your build script that we can reproduce.

bridystone, 2020-01-19:
Hi @Makuna , @proddy ,

Thanks for your fast response!

I actually did the following:
- I forked Tonuino "https://github.com/xfjx/TonUINO" and imported it to platformio
- i added all dependencies to platformio (incl. DFPlayerMp3)
- it turned out not compilable (because of some issues in Tonuino, but also in the includes).

One of the issues was, that during compilation the implementation part did not know the full declaration (because it is declared later) - changing the order did the trick.

My thinking: 
Usually you declare your stuff in Header files and implement it in cpp. I understood that with templates this is not that easy.
see: https://www.codeproject.com/Articles/48575/How-to-define-a-template-class-in-a-h-file-and-imp

So i thought this is the problem and it could be solved by changing the order.
I actually didn't create a build script - i just took the standard from platformio...

My Test-repository:
https://github.com/bridystone/TonUINO

If there would be an easy way to tell platformio to do first a declaration check and later a compile, i would be glad and it would solve the purpose.

BR
Jens

proddy, 2020-01-19:
try adding `lib_ldf_mode = chain+` to your `platformio.ini` file.

https://docs.platformio.org/en/latest/librarymanager/ldf.html#dependency-finder-mode

bridystone, 2020-01-20:
I currently have no time to test it, but it sounds like a valid solution! Thanks very much. 

I'll close the issue.

Thanks again!
Jens

bridystone, 2020-01-20:
Hi guys,

i've now checked again on the topic and found out the root cause of the "problem".
it is *not* the order of the public or private declarations.
i completely misinterpreted all the 67 error messages, by starting from top to bottom... ;)

it is actually that uint16_t and uint8_t were not defined.
And this is because... "Arduino.h" is not included 

In my project the DFPlayer.h was included first. "Arduino.h" was included at later stage.
So the compiler was not able to understand the "uint16_t".
I guess during my "fixing" i added the "arduino.h" later due to other errors and didn't recognize that this actually fixed the problem.

I've now checked on other Libraries targeting Arduino. all of them included Arduino.h in their includes. (JC_Button, MFRC221,...)
So in most cases this problem won't occur, because other libraries are included before DFPlayer. Only if DFPlayer is included first (and not used with ArduinoIDE), there will be compiler errors.

Making a long story short - My changed request to you:
- Please add "#include <Arduino.h>" to your great library

Thanks
Jens

Makuna, 2020-01-20:
By default, Arduino IDE includes Arduino.h as part of its compile (it used to require the sketch do this, but hasn't for many, many years now); so it is not needed by library header files; but maybe needed by the c/cpp files.

I guess your compile script does not do this, which means you should include it at the top of your sketch before including library header files?

BTW, these types are not Arduino types, they are standards defined in inttypes.h.  so...
```
#include <inttypes.h>
```

proddy, 2020-01-21:
if you have any files with the ,ino file extension both the Arduino IDE and PlatformIO will add the <Arduino.h> header for you.

bridystone, 2020-01-21:
ok. I just thought it would be good to add include <Arduino.h> to the beginning of your library, that such issue won't occur again.
All other libraries i've check, do this - because they use stuff in their libraries that rely on this.

If you don't plan to do this - it is ok for me.
I now know the cause of the issue and can handle this.

I'll close it again.

Thanks again for support.
Jens

closed 2020-01-21 by bridystone

Makuna, 2020-01-21:
I might still do this, but I was hoping you could confirm if just any of the two I listed above would work.  I don't use platformio myself so I can't test it.

bridystone, 2020-01-21:
Hi @Makuna ,

the forked code i'm working on is the following:
https://github.com/bridystone/TonUINO/tree/platformio_import/src
There are quite some errors in the code that Arduino IDE seems to accept - so i needed to fix more than one.

I can confirm the following:

`#include <DFMiniMp3.h>`
as first include in the central cpp-file causes a lot of errors...

Any of the following include solve the issue, if included before "DFMiniMp3.h"
`
#include <Arduino.h>
#include <DFMiniMp3.h>
...
`
`
#include <inttypes.h>
#include <DFMiniMp3.h>
...
`
`
#include <stdint.h>
#include <DFMiniMp3.h>
...
`
or even
`
#include <EEPROM.h>
#include <DFMiniMp3.h>
`
because it includes <Arduino.h>

I hope this helps
Jens

seisfeld, 2020-01-21:
I don’t know if that is particularly helpful, but there have been multiple attempts to make the TonUINO source compatible with PlatformIO by various ppl. I follow that loosely (I don’t use it myself), but I never noticed that someone had issues like the ones described here. I suggest to check under the TonUINO forks for the PlatformIO ones and see what these do differently. Maybe that gives a clue.

bridystone, 2020-01-22:
good suggestion.
i've found now found 2 forks for pio integration of Tonuino.
The relevant files are here:
https://github.com/AlexanderWillner/TonUINO/blob/cleanup/Tonuino/Tonuino.cpp
https://github.com/mutanwulf/TonUINO-PlatformIO/blob/master/include/main.h

one included <stdint.h> before DFMiniMp3 the other included <SoftwareSerial.h> before it.

hope this helps...
Jens