Makuna/DFMiniMp3 #37: Enum DfMp3_PlaySource should start at 1

ghmartin77, 2020-01-27:
```
enum DfMp3_PlaySource
{
    DfMp3_PlaySource_Usb,
```
should be:
```
enum DfMp3_PlaySource
{
    DfMp3_PlaySource_Usb = 1,
```

Please see specification.

Furthermore getCurrentTrack() has to take into account playsource set:
```
    uint16_t getCurrentTrack()
    {
        uint8_t command = 0x4c;
        
        switch (_playSource) {
          case DfMp3_PlaySource_Usb:
            command = 0x4b;
            break;
          case DfMp3_PlaySource_Sd:
            command = 0x4c;
            break;
          case DfMp3_PlaySource_Flash:
            command = 0x4d;
            break;
        }
      
        drainResponses();
        sendPacket(command);
        return listenForReply(command);
    }
```
Member _playSource
```
    DfMp3_PlaySource _playSource = DfMp3_PlaySource_Sd;
```
should be set in setPlaybackSource() like so:
```
    void setPlaybackSource(DfMp3_PlaySource source)
    {
        _playSource = source;
        sendPacket(0x09, source, 200);
    }

    DfMp3_PlaySource getPlaybackSource()
    {
        return _playSource;
    }

```

Makuna, 2020-01-27:
https://github.com/Makuna/DFMiniMp3/pull/40

Makuna, 2020-01-29:
https://github.com/Makuna/DFMiniMp3/pull/41

Makuna, 2020-02-22:
v1.0.6

closed 2020-02-22 by Makuna