Makuna/DFMiniMp3 #38: Fix/improve handling of 0x3F command/reply

ghmartin77, 2020-01-27:
listenForReply() should look like this (no "else if"):
```
                    case 0x3F:
                        if (replyArg & 0x02)
                        {
                            _isOnline = true;
                            T_NOTIFICATION_METHOD::OnCardOnline(replyArg);
                        }
                        if (replyArg & 0x01)
                        {
                            _isOnline = true;
                            T_NOTIFICATION_METHOD::OnUsbOnline(replyArg);
                        }
                        break;
```

E.g. YX5200-24SS sends 0x03 if both USB and SD are inserted. However, MH2024K-24SS sends 0x02 if both USB and SD are inserted. 

Additionally some hardware replies on 0x3F request, thus proposing to add as an enhancement:
```
    // Does not work with all models.
    // YX5200-24SS - sends reply
    // MH2024K-24SS - sends NO reply --> results in error notification
    uint8_t queryStorageDevices()
    {
        drainResponses();
        sendPacket(0x3f);
        return listenForReply(0x3f);
    }
```

This comes in handy whenever both uC and DFPlayer are powered on in parallel. In these cases you mostly miss the announcement 0x3F messages because the uC needs longer too initialize.
Otherwise you can only reset the DFPlayer to collect the 0x3F messages causing some ugly click noises.

Makuna, 2020-01-27:
what does queryStorageDevices return?

In general, it looks like you discovered that the play source value is actually a bit field and not value; so each bit represents a device rather than a series of values representing a device.  The specification is rather vague on this fact but last several of your "issues" are demonstrating this fact.

With this find, maybe a little API/notifications are required to refactor this in general.

ghmartin77, 2020-01-27:
I studied the following so-called spec-files:
http://www.picaxe.com/docs/spe033.pdf
https://github.com/PowerBroker2/DFPlayerMini_Fast/blob/master/extras/FN-M16P%2BEmbedded%2BMP3%2BAudio%2BModule%2BDatasheet.pdf

The latter one seems to be the best you can get. Most commands are documented and the common errors of the former one (e.g. mixing USB/SD ID's) are fixed.

The English translation, however is "interesting" in both :)
That's why I tried to reverse engineer by testing what I interpreted from the documents content. Chapter 3.7.1 in the second file refers to the point at hand.
0x3F is used as an unattended feedback message sent by the player shortly after powering up. Unfortunately you most likely won't see these messages due to the reason given above. I tested it by delaying supplying power to the player until the uC (Arduino in my case) was ready to receive messages.
Saw them for both test modules as mentioned above.

As chapter 3.7.1 title reads "QUERY current online storage device", I tried to send a 0x3F message to the device: YX5200-24SS replies, MH2024K-24SS does not.
Bit field can be interpreted in exactly the same way as for 0x3F messages received in listenForReply(), i.e. bit 0 is USB, bit 1 is SD card.

Makuna, 2020-01-27:
https://github.com/Makuna/DFMiniMp3/pull/40

Makuna, 2020-01-29:
https://github.com/Makuna/DFMiniMp3/pull/41

Makuna, 2020-02-22:
v.1.0.6

closed 2020-02-22 by Makuna