Makuna/DFMiniMp3 #39: Improve getTotalTrackCount taking into account USB and SD

ghmartin77, 2020-01-27:
Introduce different methods for USB and SD:
```
    uint16_t getTotalTrackCountUsb()
    {
        drainResponses();
        sendPacket(0x47);
        uint16_t result = listenForReply(0x47);
        return result;
    }

    uint16_t getTotalTrackCountSd()
    {
        drainResponses();
        sendPacket(0x48);
        uint16_t result = listenForReply(0x48);
        return result;
    }
```
For backwards-compatibility implement getTotalTrackCount() based on PlaybackSource set invoking one or the other. Please also see #37.

Makuna, 2020-01-27:
https://github.com/Makuna/DFMiniMp3/pull/40

Makuna, 2020-01-29:
https://github.com/Makuna/DFMiniMp3/pull/41

Makuna, 2020-02-22:
v.1.0.6

closed 2020-02-22 by Makuna