Makuna/DFMiniMp3 #42: SD Card Not Insterd + Power UP bug

hhgyu, 2020-02-09:
I can't speak English

change sendPacket function
``` C++
    void sendPacket(uint8_t command, uint16_t arg = 0, uint16_t sendSpaceNeeded = c_msSendSpace)
    {
        uint8_t out[DfMp3_Packet_SIZE] = { 0x7E,
            0xFF,
            06,
            command,
            00,
            static_cast<uint8_t>(arg >> 8),
            static_cast<uint8_t>(arg & 0x00ff),
            00,
            00,
            0xEF };

        setChecksum(out);

    Serial.print("millis : ");
    Serial.print(millis());
    Serial.print(", _lastSend : ");
    Serial.print(_lastSend);
    Serial.print(", _lastSendSpace : ");
    Serial.print(_lastSendSpace);
    Serial.print(", end : ");
    Serial.print((millis() - _lastSend));
    Serial.print(", _isOnline : ");
    Serial.println(_isOnline);
        // wait for spacing since last send
        while (((millis() - _lastSend) < _lastSendSpace) || !_isOnline)
        {
            // check for event messages from the device while
            // we wait
            loop();
            delay(1);
    Serial.print("millis : ");
    Serial.print(millis());
    Serial.print("_lastSend : ");
    Serial.print(_lastSend);
    Serial.print(", _lastSendSpace : ");
    Serial.print(_lastSendSpace);
    Serial.print(", end : ");
    Serial.print((millis() - _lastSend));
    Serial.print(", _isOnline : ");
    Serial.println(_isOnline);
        }

        _lastSendSpace = sendSpaceNeeded;
        _serial.write(out, DfMp3_Packet_SIZE);

        _lastSend = millis();
    }
```

log
```
Com Error 1 // 1. SD Card Not Insertd + PowerUP
Card inserted 2 // 2. SD Card Instert
//3. looped sendPacket
_lastSend : 925, _lastSendSpace : 600, end : 2503, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2504, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2509, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2514, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2520, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2526, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2532, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2537, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2543, _isOnline : 0
_lastSend : 925, _lastSendSpace : 600, end : 2549, _isOnline : 0
[... 326 lines cut ...]
                        return 0;
                        break;

                    default:
                        // unknown/unsupported command reply
                        break;
                    }
                }
            }
            else
            {
                if (replyArg != 0)
                {
                    T_NOTIFICATION_METHOD::OnError(replyArg);
                    if (_serial.available() == 0)
                    {
                        return 0;
                    }
                }
            }
        } while (command != 0);

        return 0;
    }
```

Makuna, 2020-02-09:
this code seems rather drastic and counter productive.
```
  if (!isInsertSD) {
    while (!isInsertSD) {
      waitMilliseconds(10);
    }
    mp3.reset();
    waitMilliseconds(10);
  }
```
Only call `reset` once, don't keep calling like you are doing here.

Also, it has been found some chips are sending notifications values and the code was updated to support them.  A new release has not been made with the changes, but if you get the latest from github it should improve the notification consistency.  Please compare you old sketchs to the new examples as the notification callbacks have changed.

hhgyu, 2020-02-09:
Updated library

The problem did not solve

hhgyu, 2020-02-09:
```
                    case 0x3A:
                        _isOnline = true;
                        T_NOTIFICATION_METHOD::OnPlaySourceInserted(static_cast<DfMp3_PlaySources>(replyArg));
                        break;
```

Successd log
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
_lastSend : 34, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 4, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 9, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 14, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 19, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 25, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 30, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 36, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 41, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 46, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 52, _isOnline : 1

Com Error 1
Card inserted 
_lastSend : 88, _lastSendSpace : 600, end : 2790, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 4, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 9, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 15, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 20, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 26, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 31, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 37, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 42, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 48, _isOnline : 1
```

hhgyu, 2020-02-09:
reset is Card Insertd Only One fired
```
  if (!isInsertSD) {
    while (!isInsertSD) {
      waitMilliseconds(10);
    }
    mp3.reset();
    waitMilliseconds(10);
  }
```

reset removed. error log
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
_lastSend : 34, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 4, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 9, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 14, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 19, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 25, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 30, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 36, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 41, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 46, _isOnline : 1
_lastSend : 34, _lastSendSpace : 50, end : 52, _isOnline : 1

Com Error 1
Card inserted 
_lastSend : 88, _lastSendSpace : 600, end : 2790, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 0, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 4, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 9, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 15, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 20, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 26, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 31, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 37, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 42, _isOnline : 1
_lastSend : 2878, _lastSendSpace : 50, end : 48, _isOnline : 1

Com Error 129
volume 0
_lastSend : 2929, _lastSendSpace : 50, end : 10000, _isOnline : 1
files 100
```

Makuna, 2020-02-09:
Can you describe what your problem is?  Use Google translator if you need to.
Have you tried the examples I provide and do they work?

You have two locations in your modified code that send this same debug line with no way to tell which one is sending it.  This makes it hard for someone else to even understand your problem.

`_lastSend : 925, _lastSendSpace : 600, end : 2503, _isOnline : 0`

proddy, 2020-02-09:
I've had this "card not inserted" problem before. Turned out that module needs to be powered correctly. 3v3 is too little and 5v is too much. Use a LDO to try and get to around 4.2v. This was on an ESP32 mind you.

hhgyu, 2020-02-10:
dfplayer does not work properly when power is applied without a memory card inserted.

It works properly when the power is applied with the memory card inserted.

Schallbert, 2020-02-10:
Hi,  just a little comment here: 
Instead of a low dropout voltage regulator you might want to use a standard diode that supports enough current with flat characteristics, like 1A diode 1n4007.
My player works well with 5V.  
Make sure you prepare for  a current surge. Use electrolyte cap >=100u if necessary behind diode for increased stability.

On 9 February 2020 22:09:55 CET, Proddy <notifications@github.com> wrote:
>I've had this "card not inserted" problem before. Turned out that
>module needs to be powered correctly. 3v3 is too little and 5v is too
>much. Use a LDO to try and get to around 4.2v. This was on an ESP32
>mind you.

proddy, 2020-02-10:
@Schallbert thanks for sharing, good advice. I'll add a 1n4007 as you suggested. I also ended up not using an LDO but used a 1K resistor between the Rx and Tx to get it stable with 5V and so far its been ok.

Makuna, 2020-02-12:
@hhgyu What platform are you using?

I have found a bug in SoftwareSerial for esp8266 (esp32?) that will cause problems due to bugs in the `readBytes` method.

But, I think I finally understand your problem (which you never really stated in the first place).  That is, when the SD is not present when the MP3 player is powered on, there are issues.  I can confirm this and I have some changes coming to correct this path.

Makuna, 2020-02-12:
https://github.com/Makuna/DFMiniMp3/pull/43

Makuna, 2020-02-22:
v.1.0.6

closed 2020-02-22 by Makuna