Makuna/DFMiniMp3 #47: HardwareSerial not working

gruppworks, 2020-03-16:
The compiler output is "Serial1 was not declared in this scope". I don't see where Serial1 was instantiated. Have you gotten the hardware serial version to work?

Seems related to this [Pass Serial by Reference](https://stackoverflow.com/questions/7455570/how-to-pass-serial-object-by-reference-to-my-class-in-arduino)

why i would like to get hardwareserial working: i have an issue with softwareserial and bluetooth with nrf52. I've posted [my issue here](https://forums.adafruit.com/viewtopic.php?f=57&t=163513). I'm hoping that the hardware serial will help with this. The issue is if i play a track 10 times in a row, then at random times the command seems to be ignored.

seisfeld, 2020-03-16:
I have no experience with the nrf52 but I can confirm hardware serial `Serial1` works fine on an Arduino Leonardo.

gruppworks, 2020-03-16:
Thanks, it's not the board, because it just doesn't compile in Arduino IDE. Did i miss a line of code somehow? Where is Serial1 defined other than setting up the mp3 player?:
DFMiniMp3<HardwareSerial, Mp3Notify> mp3(Serial1);

seisfeld, 2020-03-16:
Well the Board Support Package for your specific board needs to define that. Like the Leonardo one does. If I choose for example the Arduino Nano (which does not have a second hardware serial) I get the same error message as you do.

Makuna, 2020-03-16:
@gruppworks You did not mention what board you are using. 

https://www.arduino.cc/reference/en/language/functions/communication/serial/

Not all boards support more than one UART, so you may have to pick another one and the table in the above link should help you.

gruppworks, 2020-03-18:
Thanks. It seems the Adafruit nRF52 doesn't support serial1. To get it to work with Serial, i had to disconnect the Rx Tx pins during uploading a sketch. After uploading, i connected the Rx Tx pins to the DFPlayer. This works. 

I can't find documentation on Adafruit's site, tho i found this [reference on Github](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Hardware/Serial1_test/Serial1_test.ino). In the header comment is this note:
This sketch demonstrate how to use Hardware Serial1 along with
 * native USB Serial on Bluefruit nRF52840.
 * Note: Bluefruit nRF52832 does not support Serial1

The Adafruit NRF52 uses the nRF52832, so therefor it does not support Serial1.

Again, the code works fine with the Adafruit nRF52 by connecting the Rx and Tx to the DFPlayer after upload.

seisfeld, 2020-03-18:
Well if makes perfect sense that you need to use `Serial` when there is no second one. Of this this chip does not have a second one you either need to do what you did OR use soft serial.

eversonsiqueira, 2020-03-31:
Gruppworks, have you tried using software serial?

#include <SoftwareSerial.h>
SoftwareSerial softSerial(10, 11); // RX, TX
DFMiniMp3<SoftwareSerial, Mp3Notify> mp3(softSerial);

closed 2020-05-18 by Makuna