Makuna/DFMiniMp3 #48: getTotalFolderCount problem

nokiandrea, 2020-03-20:
Hi, i'm using a chinese mp3 module. This library work fine but i had a problem with this function 'getTotalFolderCount'...it is not reliable, in same cases returns values ​​other than the real ones. An example, i had 3 folders in root 01,02,03....function return 4 value, if i add a new folder in root, function return 5,ecc...also if i increment folder variable (for changing folder location), in same cases return a false value and notify com error. Is there a better solution for navigate in folders?

Makuna, 2020-03-20:
Some of the chips (there are several that span over several years and how they worked internally changed by the year code) don't work the same as others.  You may have ran into one of these inconsistencies.
This sounds like you are trying to allow an arbitrary flash setup to be used with your sketch, and when I look at the general API exposed I don't think this well supported.  An example, what if the folders were labeled "01", "03", "05", then having the count doesn't help index the folder names.

Makuna, 2020-05-19:
(stale)

closed 2020-05-19 by Makuna