Makuna/DFMiniMp3 #50: Feature request: Loop random tracks from folder

marrobHD, 2020-04-03:
Please add a feature which we are able to loop a folder an play random track in it.
Something like `loopFolderrandom(1, 1-3)`

seisfeld, 2020-04-03:
I'd consider this a high level function which one need to take care for and implement yourself in the parent sketch. This would require the library to keep track of sort of a playlist which is not it's duty - IMHO.

marrobHD, 2020-04-03:
I tried 
```
randNumber = random(1, 3);
mp3.playFolderTrack(1, randNumber)
```
but it wont work

seisfeld, 2020-04-03:
There is the play random example. You can modify that one to play a specific folder or a specific file in a folder. Then hook into `OnPlayFinished()` to play the next track.

marrobHD, 2020-04-03:
That wont work for me. I want to play a random track from only one folder and not all.

Makuna, 2020-04-03:
@marrobHD Issues are not meant to be a place to ask questions or seek general help.  It is meant to be a place to "register" an issue.  The original feature request was valid, but as stated can't be done in the library.  The chip doesn't support it and it requires knowledge that the library could never have to implement it.

If you wish support, please use the gitter channel (see readme and wiki for pointers).

closed 2020-04-03 by Makuna