Makuna/DFMiniMp3 #51: Verifying DFMiniMp3

Rich1975, 2020-05-18:


Rich1975, 2020-05-18:
Error when I try to verify this sketch....
Arduino: 1.8.12 (Windows 10), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rich\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10812 -build-path C:\Users\Rich\AppData\Local\Temp\arduino_build_860864 -warnings=none -build-cache C:\Users\Rich\AppData\Local\Temp\arduino_cache_354650 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Rich\Documents\Arduino\DFMiniMp3_h\DFMiniMp3_h.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rich\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10812 -build-path C:\Users\Rich\AppData\Local\Temp\arduino_build_860864 -warnings=none -build-cache C:\Users\Rich\AppData\Local\Temp\arduino_cache_354650 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Rich\Documents\Arduino\DFMiniMp3_h\DFMiniMp3_h.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\sketch\\DFMiniMp3_h.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\sketch\\DFMiniMp3_h.ino.cpp" -o "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\sketch\\DFMiniMp3_h.ino.cpp" -o "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\sketch\\DFMiniMp3_h.ino.cpp.o"
Compiling libraries...
Compiling core...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10812 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino\\wiring_pulse.S" -o "C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864\\core\\wiring_pulse.S.o [... cut ...] C:\\Users\\Rich\\AppData\\Local\\Temp\\arduino_build_860864" -lm
C:\Users\Rich\AppData\Local\Temp\ccMLObKb.ltrans0.ltrans.o: In function `main':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Uno.

Any suggestions to fixing this?

Schallbert, 2020-05-18:
Hi Rich,  please share your solution (as you just closed this yourself). Was it a missing #include <Arduino.h> or multiple include without guard, or anything else?

Rich1975, 2020-05-18:
Sorry, didn't mean to Close

seisfeld, 2020-05-19:
It appears you are trying to compile the library alone:

> C:\Users\Rich\Documents\Arduino\DFMiniMp3_h\DFMiniMp3_h.ino

This will not work. Please open one of the examples and try again.

Rich1975, 2020-05-19:
#include "Arduino.h"
#include <SoftwareSerial.h>
#include <DFMiniMp3.h>
#include "Adafruit_ILI9341.h" 
#include <URTouch.h>          
#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_TFTLCD.h> // Hardware-specific library
#include <SPI.h>
 
the above were all  added by including by ZIP into my library and then I tried to VERIFY and got that error....
So, in general if I'm writing my own code shouldn't I FIRST verify the code BEFORE I even CONNECT the Arduino to the PC, then connect the Arduino and UPLOAD the code?.
As you can see I am a newbee to this Arduino thing... more familiar with Vacuum tubes and Transistors Ha Ha Ha!!!

Makuna, 2020-05-19:
You still didn't include a sketch that you are compiling.

When I create a new sketch I get 
```
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
Then if you include this library, it puts the include at the top
```
#include <DFMiniMp3.h>
void setup() {
  // put your setup code here, to run once:
}
void loop() {
  // put your main code here, to run repeatedly:
}
```
This compiles fine.

Makuna, 2020-05-25:
closing due to no response and not reproduceable with valid sketch.

closed 2020-05-25 by Makuna