Makuna/DFMiniMp3 #59: template and static notifiers are a head scratcher

tcurdt, 2020-12-24:
Thanks for the library and Merry Xmas.

Maybe it's my C++ that got so rusty but why would you use the template approach and static callbacks for the notifications?
As far as I can tell this makes it very hard to escape the static scope of the notifier class.
Usage wise a normal delegation seems like a much more user friendly approach.

Maybe you could expand in the example how you imagine this to work?

To continue playing I would need to trigger the next track here:

https://github.com/Makuna/DFMiniMp3/blob/master/examples/PlayMp3/PlayMp3.ino#L45

but for me that's not available in that scope.
Thoughts?

seisfeld, 2020-12-24:
I'm far from being a c/c++ expert but this is how we do it in our project. We handle the whole playstate ourselves...

https://github.com/seisfeld/TonUINO/blob/60a75f9cb10261ba1296d2e75b43b470cbaebce4/tonuino.ino#L506-L507

https://github.com/seisfeld/TonUINO/blob/60a75f9cb10261ba1296d2e75b43b470cbaebce4/tonuino.ino#L1307 and onwards for the function...

Dunno if that is what you were looking for though (probably not ;)).

Merry Xmas!

Schallbert, 2020-12-24:
I'm escaping the static callback by using a message property along with a getter function - see here my unfinished project https://github.com/Schallbert/Tonuino_UseEncoder/blob/d46ef63dfb1776a93ee6e23ad03a5acae6b6c0aa/lib/Mp3/DfMiniMp3/DFMiniMp3_implementation.h#L80

Merry Christmas!

tcurdt, 2020-12-24:
Thanks for the input guys. 

There are various (I would call them work arounds) and I am also just using an external to escape the scope.
But from an API perspective I don't find that particular elegant (to put it nicely).

I am basically wondering if there is a particular reason the original author went this way.
Maybe I am just missing the elegant way to do this with the template approach.

But now - back to preparing the food for later :-)

closed 2020-12-24 by Makuna