Makuna/DFMiniMp3 #76: How repeat current track?

massimo2021, 2021-08-21:
Please, can you show us how to write, in an Arduino sketch, the right code  to repeat current track?
I'm not a programmer and don't understand the line from DFMiniMp3.h  _void setRepeatPlayCurrentTrack(bool repeat)_

Thanks

Massimo

Schallbert, 2021-08-21:
Hi Massimo.
First, this is not the right place for your inquiry,  the "issues" tab should only be used for bug reports etc. Please move this to "DISCUSSIONS".
Use the Wiki https://github.com/Makuna/DFMiniMp3/wiki/API-Reference#void-setrepeatplayallinrootbool-repeat
to understand which function does what. When you e.g. write 
`dfmini.setRepeatPlayCurrentTrack(true);`
And then have your song played e.g. using playFolderTrack, this song should be repeating. 
You'll need some time to experiment with your hardware, play around with function calls in this library  etc. that's the best way to learn.

massimo2021, 2021-08-21:
Well. 
Thanks for the advice  :-)

closed 2021-08-21 by massimo2021