Makuna/DFMiniMp3 #80: Com Error 3

titigafr, 2022-01-08:
Hello,
I am using a DFPlayer HM-247A (GD3200B) module and
when I run my sketch I have this error and I cannot get the total number of tracks, while playing a file is working fine .

Thanks for your help.

```
initializing...

Com Error 3
status : 0
nb de fichier : 0
nb de fichier dans le dossier 1 : 3
```

Makuna, 2022-01-09:
Are you using the Library released version of the Github version?  Github has some changes to support some of the less compatible chips that have been coming out.
There is a new option third template parameter for these chips.  

```
DFMiniMp3<HardwareSerial, Mp3Notify, Mp3ChipMH2024K16SS> DfMp3(Serial); 
```

titigafr, 2022-01-09:
I'm using platformio in VScode.
my platformio file : 
```
lib_deps = 
    https://github.com/makuna/DFMiniMp3.git
```

the serial output : 
```
Ininitializing...

Com Error 3
volume 0

Com Error 3
files 0
starting...
track 1
Ȥl���Xll�p`>�l48xhR�initializing...

Com Error 3
volume 0

Com Error 3
files 0
starting...
track 1

Com Error 3
Play finished for #1
track 2
track 3
Play finished for #3
track 1
track 2
track 3
Play finished for #3
track 1

Com Error 3
track 2
track 3
Play finished for #3
track 1
track 2
```
and the sketch : 
```
#include <SoftwareSerial.h>
#include <DFMiniMp3.h>

#define AUDIOPINTX D2
#define AUDIOPINRX D6
SoftwareSerial mySoftwareSerial(AUDIOPINTX,AUDIOPINRX); 
// forward declare the notify class, just the name
//
class Mp3Notify; 

typedef DFMiniMp3<SoftwareSerial, Mp3Notify> DfMp3;
 DfMp3 dfmp3(mySoftwareSerial);

// implement a notification class,
// its member methods will get called 
//
class Mp3Notify
{
public:
  static void PrintlnSourceAction(DfMp3_PlaySources source, const char* action)
  {
    if (source & DfMp3_PlaySources_Sd) 
    {
        Serial.print("SD Card, ");
    }
    if (source & DfMp3_PlaySources_Usb) 
    {
        Serial.print("USB Disk, ");
    }
    if (source & DfMp3_PlaySources_Flash) 
    {
        Serial.print("Flash, ");
    }
    Serial.println(action);
  }
  static void OnError(DfMp3& mp3, uint16_t errorCode)
  {
    // see DfMp3_Error for code meaning
    Serial.println();
    Serial.print("Com Error ");
    Serial.println(errorCode);
  }
  static void OnPlayFinished(DfMp3& mp3, DfMp3_PlaySources source, uint16_t track)
  {
    Serial.print("Play finished for #");
    Serial.println(track);  
  }
  static void OnPlaySourceOnline(DfMp3& mp3, DfMp3_PlaySources source)
  {
    PrintlnSourceAction(source, "online");
  }
  static void OnPlaySourceInserted(DfMp3& mp3, DfMp3_PlaySources source)
  {
    PrintlnSourceAction(source, "inserted");
  }
  static void OnPlaySourceRemoved(DfMp3& mp3, DfMp3_PlaySources source)
  {
    PrintlnSourceAction(source, "removed");
  }
};

void setup() 
{
  Serial.begin(9600);
  Serial.println("initializing...");
  dfmp3.begin();
  uint16_t volume = dfmp3.getVolume();
  Serial.print("volume ");
  Serial.println(volume);
  dfmp3.setVolume(15);
  
  uint16_t count = dfmp3.getTotalTrackCount(DfMp3_PlaySource_Sd);
  Serial.print("files ");
  Serial.println(count);
  
  Serial.println("starting...");
}

void waitMilliseconds(uint16_t msWait)
{
  uint32_t start = millis();
  
  while ((millis() - start) < msWait)
  {
    dfmp3.loop(); 
    delay(1);
  }
}

void loop() 
{
  Serial.println("track 1"); 
  dfmp3.playFolderTrack(1,1);
  
  waitMilliseconds(5000);
  
  Serial.println("track 2"); 
  dfmp3.playFolderTrack(1,2);
  
  waitMilliseconds(5000);
  
  Serial.println("track 3");
  dfmp3.playFolderTrack(1,3);
  
  waitMilliseconds(5000); 
}

```

this sketch is the one that is example in github,

I tried with this line :  typedef DFMiniMp3<SoftwareSerial, Mp3Notify, Mp3ChipMH2024K16SS> DfMp3;
then this one :  typedef DFMiniMp3<SoftwareSerial, Mp3Notify, Mp3ChipOriginal> DfMp3;
but nothing changes, the error is always there.

thanks for your help.

Makuna, 2022-01-10:
If you look through the closed issues, you will see that this command is one of the problematic ones where some hardware just doesn't support it well.  Here is the query to see the closed issues.  Note the one about the filenames; maybe this is what you are hitting?

https://github.com/Makuna/DFMiniMp3/issues?q=is%3Aissue+is%3Aclosed+getTotalTrackCount

Makuna, 2022-04-09:
from a discussion, there seems to be required delays that were not present before.  The Repo has some improvements here.

I also noted that you are not calling .reset() after .begin().

Makuna, 2023-03-23:
Please pick the latest on GitHub (https://github.com/Makuna/DFMiniMp3/pull/112) as I improved the comms.  Also, if you add the following to the top of your sketch (above where you include the library) this will dump the low level communications bytes so we can see more details as to what is happening.

```
#define DfMiniMp3Debug Serial
```

Makuna, 2023-03-28:
No response for over a year, considering this solved and closing.

closed 2023-03-28 by Makuna