Makuna/DFMiniMp3 #102: "Card Online" instead of playing

kopierreko, 2022-11-18:
Hi,

I have a problem with my module... When I play a file, instead of playing the file, I receive a notification "Card Online"... I tried many SD Cards and I read somewhere that it exists a firmware upgrade of the DFR0299, so I made it. Each time I make it before restarting the uC and putting a SDCard with sounds, it works. But if I unplug/replug the 5V, Impossible. 

Do you have an idea ? 
Thanks 😊

kopierreko, 2022-11-18:
> The only thing I can think of is that you are not waiting after startup for the notifications that the device is ready for commands. I forget the exact notifications that are sent, but Card Online is one of them. You can call isOnline() to check that it is ready.
> Did you try the examples to make sure its not your sketch?
> I have never heard about a firmware updates.

Yes, I tested with the example and it doesn’t work… I tested a more simple code which works but with less functionalities, if you want to have a look.

https://drive.google.com/file/d/1DarKSM06ZiGLN_95APHyuZoIfex8JJbD/edit

kopierreko, 2022-11-18:
> What are the names of the files on your SD card?

Like that

Makuna, 2022-11-18:
Audio Files, the ones you are trying to play.

kopierreko, 2022-11-18:
It’s the content of my SD card ^^

And in my code I just test with dfmp3.playMp3FolderTrack(1);  // sd:/mp3/0001.mp3

With your lib, I can get volume, EQ, etc, receive notifications, but when I run the playMp3[…] code, nothing append and I receive the notification as if the SD cards was being inserted…

I can’t understand. I don’t see major difference btw your lib and the sample code I sent bellow which works in my case

Makuna, 2022-11-18:
That ZIP file didn't contain any audio files to play, so if you call `playMp3FolderTrack(1)` nothing will be played.  I suspect you have audio files, but you may have misinterpreted what I am asking. I am looking for the names of the files; in this case, under the mp3 folder.  I don't remember if case is important for the mp3 folder name.

Also, are you using `Mp3ChipMH2024K16SS` or `Mp3ChipOriginal` as the variant with my library?  

The big difference is that the sketch you pointed me at doesn't seem to read the status from the player.  I didn't look too deeply though.  If you want someone to inspect code, minimize the code the smallest code that reproduces the problem and include it here.  (see markdown to provide "code" formatting).

kopierreko, 2022-11-18:
I edited my previous message, I just saw that the screenshot was not visible. My code is your example code with just the port edited, not else :/

Makuna, 2023-03-23:
I would suggest two things to help track down the issue.
* pick up the latest from the GitHub as it includes some communications improvements
* add the following in your sketch at the very top before you include this library
```
#define DfMiniMp3Debug Serial
```
This will dump the low level information for each command and response and should give us a hint.

patrickzzz, 2024-07-01:
Hi @Makuna,
in one project where this library is used, i have one sd card which works, and another older one, that makes a few problems..
I encountered, that it must be due to a longer loading time of the older sd card (time that dfplayer needs to be ready) - so the reset(false) is sent but somehow not already registered by the dfplayer, and so the isOnline will not get true..

also, as not always a dfplayer is present, its not possible to wait for ever that dfplayer is available (like waitIsOnline flag)..

although: the dfplayer is autoplaying and sending info about the played songs, but this doesnt set isOnline.. would it be possible somehow, to expand the events/notifications, which will set isOnline true?

Regards,
Patrick

Makuna, 2024-07-01:
@patrickzzz The notification callback method is `OnPlaySourceOnline`. The chips vary a little as to what this even means and when it means its online.  It's one of the inconsistency areas of these chips, even on the original.  Right now it is set "online" when any notification of a play source is inserted or removed, and of course an online message.