Makuna/DFMiniMp3 #120: faulty package at platform IO

kaktus42, 2023-04-15:
I installed the library via PlatformIO (https://registry.platformio.org/libraries/makuna/DFPlayer%20Mini%20Mp3%20by%20Makuna, using `makuna/DFPlayer Mini Mp3 by Makuna@^1.2.0`) and had to realize that it's not the current version of the code. I found it from the fact that the function `void begin(int8_t rxPin, int8_t txPin, unsigned long baud = 9600)` was missing.

I don't know how you do the deployment to there, but maybe you should double check.
For now I use the source from the github release (in that tar.gz it's still correct) in the projects lib folder.

Makuna, 2023-04-15:
What version were you seeing?  You can look at the root of the library location and read it from the library.json file.

kaktus42, 2023-05-19:
sorry for the late reply.
There is no `library.json`, just a `library.properties`.

[image]

Makuna, 2023-05-19:
My Github does have both and they state they use a scraper so it should be updating.  You may want to contact them that a library is not being updated by their scraper.