Makuna/DFMiniMp3 #127: Mp3ChipMH2024K16SS does not start the next song from the beginning

serkansatuk, 2023-05-25:
Hello.
I have 3 different chip models of this DFplayer.
1. YX5200 24 pins
2. GD3200D 16 pins
3. MH2024K 24 pins

The codes work very well in the first 2 modules. but when I want to change the song in the MH2024K 24-pin model, for example when I switch from USB to SD card, instead of playing the new song, it continues to play the previous song on the SD card from where it left off. The same happens when switching to USB. I also used the MP3 stop command but the problem persists. Can you help me?

```
// start the first track playing
  Serial.println("DfMp3_PlaySource_Sd...");
  dfmp3.setPlaybackSource(DfMp3_PlaySource_Sd);
  delay(100);
  dfmp3.playFolderTrack(1,5);  // sd:/mp3/0001.mp3
  delay(10000);
  dfmp3.stop();
  delay(100);

  Serial.println("DfMp3_PlaySource_Usb...");
  dfmp3.setPlaybackSource(DfMp3_PlaySource_Usb);
  delay(100);
  dfmp3.playFolderTrack(1,2);
  delay(10000);
  dfmp3.stop();
  delay(100);

  Serial.println("DfMp3_PlaySource_Sd...");
  dfmp3.setPlaybackSource(DfMp3_PlaySource_Sd);
  delay(100);
  dfmp3.playFolderTrack(2,5);  // sd:/mp3/0001.mp3
  delay(10000);
  dfmp3.stop();
  delay(100);

  Serial.println("DfMp3_PlaySource_Usb...");
  dfmp3.setPlaybackSource(DfMp3_PlaySource_Usb);
  delay(100);
  dfmp3.playFolderTrack(1,1);
  delay(10000);
```

Makuna, 2023-05-25:
"Can you help me" are better asked in the discussions Q&A area.

https://github.com/Makuna/DFMiniMp3/discussions/122 is tracking a new chip that is **NOT** following the standards.  This discussion includes a sketch that does a comprehensive API call with debug information that allows for better diagnostics.

Makuna, 2023-05-26:
BTW, where did you get your module that supports USB and SD?