Makuna/DFMiniMp3 #148: Timout of 900ms not enough for the chip MH2024K-24SS (FD Player MP3-TF-16P V3.0)

boerge1, 2023-11-08:
With the DF Player MP3-TF-16P V3.0 with the cip MH2024K-24SS (bought in October from  AZ-Delivery) on some folder the request getFolderTrackCount() takes more than 900ms: 

```
call getFolderTrackCount()
IN  7e ff 06 3d 00 00 12 fe ac ef
OUT  7e ff 06 4e 00 00 02 fe ab ef
OUT  7e ff 06 4e 00 00 02 fe ab ef (retry after 900ms)
IN  7e ff 06 4e 00 00 0f fe 9e ef (reply for the first getFolderTrackCount request)
end getFolderTrackCount()
call playFolderTrack()
OUT  7e ff 06 0f 01 02 0b fe de ef
IN  7e ff 06 41 00 00 00 fe ba ef
IN  7e ff 06 4e 00 00 0f fe 9e ef (reply for the second getFolderTrackCount request)
OUT  7e ff 06 0d 01 00 00 fe ed ef
IN  7e ff 06 41 00 00 00 fe ba ef
here it plays not the track that is requested but the very first mp3 track on sd card. 
```

We tried to increase the timeout to 1500ms and then we had no retry anymore. 

So I would propose one of the following solutions: 

* increase the default timeout to 1500ms
* make the timeout configurable

Makuna, 2023-11-08:
I am thinking that exposing a template argument for this timeout will work, applying a custom timeout as needed.  For you, the definition of the object would look like
```
DFMiniMp3<HardwareSerial, Mp3Notify, Mp3ChipMH2024K16SS, 1500> dfMp3(Serial);
```
Where the last optional template argument is the timeout in ms; 1500 in this case.  Since its optional its default will be the normal 900ms.

Sound good?

Makuna, 2023-11-08:
[see branch CustomTimeout](https://github.com/Makuna/DFMiniMp3/tree/CustomTimeout)

boerge1, 2023-11-09:
Yes, it sounds very good! 
But we would have to use 
`DFMiniMp3<HardwareSerial, Mp3Notify, Mp3ChipOriginal, 1500> dfMp3(Serial);`

Makuna, 2023-11-09:
Yes, did you try it?  Just clone that branch into the library folder to try it.

Once it works for you, I will merge the changes into Master and create a new release.

boerge1, 2023-11-12:
Yes, it works for us. So you can merge it and it would be fine if you create a release.
Many thanks!

boerge1, 2023-11-19:
Do you already have a plan when it will be released?

Makuna, 2023-11-19:
I was waiting on another discussion to be resolved in case there is something else to add.
Probably mid week at the latest.

Makuna, 2023-11-20:
https://github.com/Makuna/DFMiniMp3/releases/tag/1.2.3

closed 2023-11-20 by Makuna