Makuna/DFMiniMp3 #165: c_NoAckTimeout too short

patheo, 2024-04-02:
Hi, First of all, thanks for your work !! 
I'm working on a new project using Esp8266 D1 mini and DFPlayer (MH2024K-24SS). I find that the c_NoAckTimeout looks too short. With the default value (50ms), esp8266 stop send data to DFPlayer after an error (error 6 - can't find file). With 500ms it works very great !
So it's not really an issue, more like a quick note ;-)
Thanks

Makuna, 2024-04-02:
How are you defining your Mp3 object?  
What method are you calling that returns the error?
Non-Issues should be started in the Discussions area, not the issues area.

I think you are mixing up a "request error" with an async reply from the chip.  Its subtle at times especially with chips that don't support an Ack feedback.

A request error is when the chip doesn't understand what you are requesting, or the current state doesn't allow that.  These are immediate.  Its either an ACK or an error.

A request that requires processing before it can reply, like file not found, sd read fail; aren't a reply to the command and are more like the file finishes playing notifications. 
These are sent asynchronously, and the timing is highly variable and depends on how many files and folders you have.

patheo, 2024-04-04:
Hi, thanks for your answer !

I define the Mp3 object like that :
```
SoftwareSerial secondarySerial(D3, D4);
typedef DFMiniMp3<SoftwareSerial, Mp3Notify, Mp3ChipIncongruousNoAck, 1600> DfMp3;
DfMp3 dfmp3(secondarySerial);
```

and I play file with this function :

`dfmp3.playFolderTrack(1, 1);`

If the file exists, everything works great. But if the file doesn't exist, I receive immediately the error 6 message (absolutely normal) but when I try to call a new file (existing one), nothing happened. 
I try to call `dfmp3.reset(); ` just after an error arrived and it works (I can hear the speaker rebooting), but I can't play a new file.

Changing the c_NoAckTimeout value to 500ms (just added a 0) solved the problem.

Thanks for your help.

Makuna, 2024-04-04:
So wait, it's not about the first failed call?  

"if the file doesn't exist, I receive immediately the error 6 message"

So, this error returns within 50ms or 500ms?

"but when I try to call a new file (existing one), nothing happened"

No errors, nothing played?  Are you calling this before the error returns?

Try to avoid calling reset(), some chips get into a weird state when it's called.  
The NoAck is a problematic chip design.  Your solution means EVERY single call will wait 500ms even if it doesn't need to, waiting for a reply of some sort.  The real issue isn't the c_NoAckTimeout is too short, it's something else as changing these masks the core problem.

patheo, 2024-04-04:
_So wait, it's not about the first failed call?_ 
If the first call is an existing file, nothing fail.

_So, this error returns within 50ms or 500ms?_
This error return within 50ms before I changed the `c_NoAckTimeout` value.

_No errors, nothing played? Are you calling this before the error returns?_
Exactly, no errors but nothing played. Like the serial communication stopped. For information, the esp8266 loop don't stop after that. I call the new file after receiving the error message.

_Try to avoid calling reset()_
Same problem with and without the reset() call...

_The real issue isn't the c_NoAckTimeout is too short, it's something else as changing these masks the core problem._
Ok, I understand that, thanks for the explanation (I'm quite a newbie), but I don't know how find the solution, or help you find it.

Thanks again for your help

Makuna, 2024-04-05:
Next step.  [Follow this](https://github.com/Makuna/DFMiniMp3/wiki/Debug-Features#dfminimp3debug) to add debug output support.  Make sure to turn "show timestamps" in the serial monitor.  

Revert your change to ack timeout.

Then capture the debug output when demonstrating the problem.

patheo, 2024-04-08:
Thanks for your answer. I'm not at home until Thursday. I'll try this at this moment !

patheo, 2024-04-11:
This is the result of the test :

11:36:25.166 -> initializing...
11:36:25.166 -> SD Card, online
11:36:25.198 -> starting...
11:36:25.198 -> OUT  7e ff 06 46 00 00 00 fe b5 ef
11:36:26.019 -> OUT  7e ff 06 46 00 00 00 fe b5 ef
11:36:26.938 -> OUT  7e ff 06 46 00 00 00 fe b5 ef
11:36:27.859 -> version 0
11:36:27.859 -> playGlobalTrack 6
11:36:27.859 -> OUT  7e ff 06 03 01 00 06 fe f1 ef
11:36:28.913 -> playMp3FolderTrack 2
11:36:28.913 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:28.946 -> OUT  7e ff 06 12 01 00 02 fe e6 ef
11:36:28.978 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:29.966 -> playFolderTrack 20, 2
11:36:29.966 -> IN  7e ff 06 40 00 00 06 fe b5 ef
11:36:29.999 -> OUT  7e ff 06 0f 01 14 02 fe d5 ef
11:36:31.050 -> playFolderTrack16 10, 2000
11:36:31.050 -> 
11:36:31.050 -> Com Error 6
11:36:31.083 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:31.117 -> IN  7e ff 06 40 00 00 06 fe b5 ef
11:36:31.149 -> OUT  7e ff 06 14 01 a7 d0 fd 6f ef
11:36:32.102 -> playRandomTrackFromAll
11:36:32.102 -> 
11:36:32.102 -> Com Error 6
11:36:32.168 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:32.202 -> IN  7e ff 06 40 00 00 06 fe b5 ef
11:36:32.234 -> OUT  7e ff 06 18 01 00 00 fe e2 ef
11:36:33.182 -> nextTrack
11:36:33.182 -> 
11:36:33.182 -> Com Error 6
11:36:33.182 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:33.214 -> OUT  7e ff 06 01 01 00 00 fe f9 ef
11:36:33.280 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:34.267 -> prevTrack
11:36:34.267 -> OUT  7e ff 06 02 01 00 00 fe f8 ef
11:36:34.333 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:35.319 -> OUT  7e ff 06 4c 00 00 00 fe af ef
11:36:35.383 -> IN  7e ff 06 4c 00 00 2a fe 85 ef
11:36:35.415 -> current track 42
11:36:35.415 -> setVolume 16
11:36:35.448 -> OUT  7e ff 06 06 01 00 10 fe e4 ef
11:36:35.513 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:35.546 -> OUT  7e ff 06 43 00 00 00 fe b8 ef
11:36:35.578 -> IN  7e ff 06 43 00 00 10 fe a8 ef
11:36:35.611 -> volume 16
11:36:35.611 -> increaseVolume
11:36:35.644 -> OUT  7e ff 06 04 01 00 00 fe f6 ef
11:36:35.677 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:35.710 -> decreaseVolume
11:36:35.743 -> OUT  7e ff 06 05 01 00 00 fe f5 ef
11:36:35.776 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:35.807 -> loopGlobalTrack 6
11:36:35.807 -> OUT  7e ff 06 08 01 00 06 fe ec ef
11:36:35.874 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:36.794 -> loopFolder 20
11:36:36.794 -> OUT  7e ff 06 17 01 00 14 fe cf ef
11:36:36.860 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:37.881 -> setPlaybackMode 2 (single repeat)
11:36:37.913 -> IN  7e ff 06 40 00 00 06 fe b5 ef
11:36:37.945 -> OUT  7e ff 06 08 01 00 02 fe f0 ef
11:36:37.979 -> 
11:36:37.979 -> Com Error 6
11:36:37.979 -> OUT  7e ff 06 45 00 00 00 fe b6 ef
11:36:38.043 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:38.076 -> OUT  7e ff 06 45 00 00 00 fe b6 ef
11:36:38.141 -> IN  7e ff 06 45 00 00 02 fe b4 ef
11:36:38.174 -> playback mode 2
11:36:38.174 -> setRepeatPlayAllInRoot false
11:36:38.207 -> OUT  7e ff 06 11 01 00 00 fe e9 ef
11:36:39.225 -> setRepeatPlayCurrentTrack true
11:36:39.258 -> OUT  7e ff 06 19 01 00 00 fe e1 ef
11:36:39.292 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:40.280 -> setEq 3 (jazz)
11:36:40.280 -> OUT  7e ff 06 07 01 00 03 fe f0 ef
11:36:40.346 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:40.378 -> OUT  7e ff 06 44 00 00 00 fe b7 ef
11:36:40.411 -> IN  7e ff 06 44 00 00 03 fe b4 ef
11:36:40.445 -> eq mode 3
11:36:40.445 -> setPlaybackSource 2 (SD)
11:36:40.477 -> OUT  7e ff 06 09 01 00 02 fe ef ef
11:36:40.510 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:41.465 -> start
11:36:41.465 -> OUT  7e ff 06 0d 01 00 00 fe ed ef
11:36:41.530 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:42.552 -> pause
11:36:42.552 -> OUT  7e ff 06 0e 01 00 00 fe ec ef
11:36:42.618 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:43.602 -> stop
11:36:43.602 -> OUT  7e ff 06 16 01 00 00 fe e4 ef
11:36:44.685 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:44.719 -> OUT  7e ff 06 42 00 00 00 fe b9 ef
11:36:44.783 -> IN  7e ff 06 42 00 00 00 fe b9 ef
11:36:44.815 -> status source is 0 and state is 0
11:36:44.847 -> OUT  7e ff 06 4e 00 00 0a fe a3 ef
11:36:45.629 -> IN  7e ff 06 4e 00 00 00 fe ad ef
11:36:45.661 -> folder 10 files 0
11:36:45.661 -> OUT  7e ff 06 48 00 00 00 fe b3 ef
11:36:45.760 -> IN  7e ff 06 48 00 00 79 fe 3a ef
11:36:45.794 -> total files 121
11:36:45.794 -> OUT  7e ff 06 4f 00 00 00 fe ac ef
11:36:46.647 -> OUT  7e ff 06 4f 00 00 00 fe ac ef
11:36:47.569 -> OUT  7e ff 06 4f 00 00 00 fe ac ef
11:36:48.489 -> total folders 0
11:36:48.489 -> playAdvertisement 2
11:36:48.521 -> OUT  7e ff 06 13 01 00 02 fe e5 ef
11:36:49.539 -> stopAdvertisement
11:36:49.539 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:49.572 -> OUT  7e ff 06 15 01 00 00 fe e5 ef
11:36:49.605 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:50.624 -> enableDac
11:36:50.624 -> OUT  7e ff 06 1a 01 00 00 fe e0 ef
11:36:50.690 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:50.723 -> disableDac
11:36:50.723 -> OUT  7e ff 06 1a 01 00 01 fe df ef
11:36:50.757 -> IN  7e ff 06 41 00 00 00 fe ba ef
11:36:50.791 -> ...stopping

Makuna, 2024-04-11:
reviewed the first few requests.  See annotated below.

First, your chip does support ack, so don't use `Mp3ChipIncongruousNoAck`.
Second, your chip is a little slow to respond sometimes, but you are already setting it to 1600 so that shouldn't be an issue.  The default of 900, which is not enough for your chip.  So your good here.
If you spam requests at the chip before it can ack/fail, you can put it into a poor state.  Since you set it to ignore the acks, it is allowing you to send commands far faster than you should.

Switch to `Mp3ChipOriginal` or `Mp3ChipMH2024K16SS`.

```
11:36:25.166 -> initializing...
11:36:25.166 -> SD Card, online
11:36:25.198 -> starting...
11:36:25.198 -> OUT 7e ff 06 46 00 00 00 fe b5 ef	Mp3_Commands_GetSoftwareVersion
11:36:26.019 -> OUT 7e ff 06 46 00 00 00 fe b5 ef
11:36:26.938 -> OUT 7e ff 06 46 00 00 00 fe b5 ef
11:36:27.859 -> version 0			never replied, so it doesn't support version command
11:36:27.859 -> playGlobalTrack 6
11:36:27.859 -> OUT 7e ff 06 03 01 00 06 fe f1 ef	Mp3_Commands_PlayGlobalTrack
11:36:28.913 -> playMp3FolderTrack 2
11:36:28.913 -> IN 7e ff 06 41 00 00 00 fe ba ef	Mp3_Replies_Ack (after 1054ms)
11:36:28.946 -> OUT 7e ff 06 12 01 00 02 fe e6 ef	Mp3_Commands_PlayMp3FolderTrack (only after 1087ms of requesting playing global track)
11:36:28.978 -> IN 7e ff 06 41 00 00 00 fe ba ef	Mp3_Replies_Ack (after 32ms)
11:36:29.966 -> playFolderTrack 20, 2
11:36:29.966 -> IN 7e ff 06 40 00 00 06 fe b5 ef                Mp3_Replies_Error - DfMp3_Error_FileMismatch
11:36:29.999 -> OUT 7e ff 06 0f 01 14 02 fe d5 ef	Mp3_Commands_PlayFolderTrack
11:36:31.050 -> playFolderTrack16 10, 2000
11:36:31.050 ->
11:36:31.050 -> Com Error 6			DfMp3_Error_FileMismatch
11:36:31.083 -> IN 7e ff 06 41 00 00 00 fe ba ef	Mp3_Replies_Ack
11:36:31.117 -> IN 7e ff 06 40 00 00 06 fe b5 ef	Mp3_Replies_Error - DfMp3_Error_FileMismatch
11:36:31.149 -> OUT 7e ff 06 14 01 a7 d0 fd 6f ef	Mp3_Commands_PlayFolderTrack16
11:36:32.102 -> playRandomTrackFromAll
11:36:32.102 ->
11:36:32.102 -> Com Error 6			DfMp3_Error_FileMismatch
11:36:32.168 -> IN 7e ff 06 41 00 00 00 fe ba ef	Mp3_Replies_Ack
11:36:32.202 -> IN 7e ff 06 40 00 00 06 fe b5 ef	Mp3_Replies_Error - DfMp3_Error_FileMismatch
11:36:32.234 -> OUT 7e ff 06 18 01 00 00 fe e2 ef	Mp3_Commands_PlayRandmomGlobalTrack
11:36:33.182 -> nextTrack
```

patheo, 2024-04-19:
Thanks for your answer ! Everythings works great with this :
`typedef DFMiniMp3<SoftwareSerial, Mp3Notify, Mp3ChipOriginal, 1600> DfMp3;`